AmberELEC/AmberELEC #532: MS-DOS library limits extensions to "*.dosz" files

joelteixeira, 2021-07-05:
Hey there, as discussed with acridAxid on "general" channel from discord, the actual implementation of the MS-DOS library limits the listing to only "*.doz" extensions.  This is valid for DOSBOX Pure, but, if the user, for whatever reason, wants to use another core "standard dosbox" for instance it won't be possible to launch other games since "standard dosbox" doesn't read  dosz files and will drop the user to a command prompt with error. 

Actual behaviour: 

Only dosz files are listed. 

Expected behaviour: 

-   List all dosz files
-   List all folders inside /roms/pc
-   List all EXE, BAT, COM files.  

Steps to reproduce:

1 - Take any game on MS-DOS library
2 - Press X to game options
3 - Advanced game options
4 - Change emualator from dosbox pure to dosbox-svn. 
5 - When opening the user will be dropper to command prompt. 

Scenario: 

When trying to play Fallout, the mouse cursor on dosbox pure is extremely laggy what turns the game unplayable. On traditional dosbox the cursor transition is much more smooth.

dhwz, 2021-07-05:
I know we will change that somewhere in the future.

SuperBadger77, 2021-07-05:
That would be a nice feature to have.

Rolen47, 2021-07-06:
If you compress Fallout with 0 compression does that make a difference?

joelteixeira, 2021-07-06:
Hi @Rolen47 it shouldn’t make any difference since it would need to unpack the archive (even without compression) to be able to point to the specific executable file.

Rolen47, 2021-07-06:
Dosbox Pure doesn't decompress the entire archive.  It streams the individual files and decompress them in real-time as it needs them.

Rolen47, 2021-07-06:
From what I read Dosbox pure doesn't completely decompress the archive it seeks the files inside the zip and decompresses the individual files in real time so I thought maybe using no compression would use less CPU. But I just tried Fallout with 0 compression and it has the same performance in DosBox Pure. The DosBox Pure core just has worse performance compared to the Dosbox SVN core.

ffffyyc, 2021-07-07:
I have revised “es_systems.cfg” to make .bat working.

joelteixeira, 2021-07-07:
@ffffyyc 

Can you detail how you did it? I followed the link you pointed and end up with the following file and content. I have a AAAAA.bat file on the folder but it stills unlisted. 

```
~/.config # cat es_system_bat.cfg
<systemList>
  <system>
    <name>bat</name>
     <path>/storage/roms/pc</path>
  </system>
</systemList>
```

ffffyyc, 2021-07-07:
> @ffffyyc
>
> Can you detail how you did it? I followed the link you pointed and end up with the following file and content. I have a AAAAA.bat file on the folder but it stills unlisted.
>
> ```
> ~/.config [...]

Before you create a custom “es_system.cfg” file. 
you should check original file located at “/usr/config/emulationstation/es_systems.cfg”
that you can find how “es_system.cfg” works .

dhwz, 2021-07-18:
fixed

closed 2021-07-18 by dhwz