AmberELEC/AmberELEC #538: Advanced System Options is not setting the correct values for some Gameboy Colorization options.

Rolen47, 2021-07-08:
### Describe the bug
The colorization option for "GBC" and "SGB" is not an "internal" palette, Advanced System Options is setting them incorrectly.

### To Reproduce
Before selecting an original Gameboy game press select and to go to Advanced System Options
Colorization
Select SGB or GBC.
Load game (in my test I used Pokemon Blue)
Check Colorization settings, they are wrong

Advanced System Options sets the following settings incorrectly:
Auto:
gambatte_gb_colorization = "disabled"
gambatte_gb_internal_palette = ""

GBC:
gambatte_gb_colorization = "internal"
gambatte_gb_internal_palette = "GBC"

SGB:
gambatte_gb_colorization = "internal"
gambatte_gb_internal_palette = "SGB"

### Expected behavior
This is the correct settings that Advanced System Options should be setting for these options:
Auto:
gambatte_gb_colorization = "auto"
gambatte_gb_internal_palette = ""

GBC:
gambatte_gb_colorization = "GBC"
gambatte_gb_internal_palette = ""

SGB:
gambatte_gb_colorization = "SGB"
gambatte_gb_internal_palette = ""

The rest of the colorization options appear to be functioning correctly because they are "Internal" colorization.

### Device (please complete the following information):
RG351V with 351ELEC 20210603

dhwz, 2021-07-08:
@konsumschaf can you please check that

konsumschaf, 2021-07-16:
Start working on this ...

konsumschaf, 2021-07-16:
Fixed in https://github.com/351ELEC/351ELEC/pull/556

closed 2021-07-16 by konsumschaf