AmberELEC/AmberELEC #560: Battery icon is Wrong

DiogoCoelhoWeb, 2021-07-18:
I'm having a problem with the battery icon. The bug is that it shows the charging icon even when the console is not charging.

Idk if this happens to everyone but to me it happens every time when the device is running.

[image]

System details - RG351P running 351Elec V20210603

bd30, 2021-07-18:
Related issue - RG351P running 351Elec V20210603

The battery icon shows a high charge (e.g. 100%, 90%) even when the battery is actually much lower. Multiple times it has said "100%" for over an hour.

DiogoCoelhoWeb, 2021-07-19:
I haven't noticed that problem on mine but I'll keep an eye on it.

pkegg, 2021-07-19:
So - we end up getting that information directly from the linux kernel.  Likely, this is a hardware issue and there is nothing we can do, but it is possible Anbernic included some sort of different hardware in your version of the RG351P that we don't account for, etc.

If you have a wifi dongle and can ssh to your device (user: root password: 351elec), run this command when the issue is happening, that could help us figure out what is going on.  On windows, you can use PuTTY, etc.

If you can ssh to the device, after logging in, type: `cat /sys/class/power_supply/battery/status` when it's unplugged and showing this issue.  Then wait about 10 seconds after plugging in, run the command again and see if the output is different.

For example, this is what it looks like this for me (device unplugged, run command, plug in and wait 10 seconds, then run again):
```
~ # cat /sys/class/power_supply/battery/status 
Discharging
~ # cat /sys/class/power_supply/battery/status 
Charging
```

DiogoCoelhoWeb, 2021-07-19:
I can see battery percentage fluctuations (changer in: 94% charger out: 74%) and the chrging light works as expected so it's probably a kernel bug ... Do have any ideas on how to fix this?

Thanks for helping

pkegg, 2021-07-20:
> I can see battery percentage fluctuations (changer in: 94% charger out: 74%) and the chrging light works as expected so it's probably a kernel bug ... Do have any ideas on how to fix this?

The battery percentage is basically not accurate on alll RG351 devices and there's not anything we can do about it.  What I do is just turn off the battery percentage text in ES.  This doesn't really 'fix' anything, but since the batter meter has to be off by 20+% to cause a change, it's less aggravating :)

> It says charging both when the charger is pulgged and unplugged.

Yeah - that's almost certainly a hardware issue then - it could also be something in the kernel - but given it seems only to be happening for you that seems somewhat unlikely.  Even if it **was** in the kernel, none of the kernel was written by the 351ELEC team, so we can't easily fix it anyway.

I'm going to close this issue as I don't think there's anything we can do about it and is almost certainly a hardware issue :(.  Feel free to join us on discord if you need additional support, etc.

closed 2021-07-20 by pkegg

DiogoCoelhoWeb, 2021-07-20:
I don´t think that it is an hardware issue, because it worked on the original firmware and on 351Elec for quite a while and only then stopped wroking. The device as been taken really good care so I find it difficult to belive that the problem is hardware ... I think it might be kernel you said earlier

sertiana, 2021-07-20:
There is nothing the devs can do in the software side to make the battery percentage more accurate. We just need to deal with it.