AmberELEC/AmberELEC #567: GBA - Cracking sound after sleep

Sichroteph, 2021-07-21:
Currently on mgba. After sleep there is a little cracking sound, and the fps lost one frame. 
Leaving RetroArch and launching the game again fix it. 
Current game : a link to the past, GBA port.

Sichroteph, 2021-07-21:
Edit : I switched smooth game ON and the issue disappeared, even after switching the option OFF

RupeeClock, 2021-07-23:
I experience similar issues on my RG351V when playing SNES games.
This is on the Crazy Hedgehog release (not sure if hotfixed or not).
Retroarch 1.9.4, Snes9x (1.60 63e20af3).

I frequently play Super Metroid + A Link to the Past combo randomizer. When resuming the game from sleep, the cracking sound becomes prominent particular in the Super Metroid side of the game.
ALttP is LoROM whilst Super Metroid is HiROM, the latter has a higher performance requirement thus the crackling becomes more noticeable.

I play a lot of GBA games on my RG351P with the same firmware but have not personally experienced this sound issue. I mostly play Mega Man Battle Network games on it.

GibzFGC, 2021-07-23:
Sleep mode on these devices causes all kinds of strange issues and it's hardware related so software can't fix it unfortunately

RupeeClock, 2021-07-23:
That's unfortunate. A workaround does at least exist to re-launch Retroarch, which works well if you use the auto-savestate option for quick resume.

sertiana, 2021-07-23:
Using auto save/load when waking up from sleep has been reported to corrupt actual saves, so that's still not a good solution unfortunately.

closed 2021-07-23 by dhwz