AmberELEC/AmberELEC #578: no gamepad detected, hold a button on your device to configure

johnnyzen, 2021-07-27:
I have installed the following images:

351ELEC-RG351P.aarch64-20210422.img
351ELEC-RG351P.aarch64-20210603.img

Device: RG351M purchased, this month.

After boot, everything seems to run fine, but then I get the following message:

"no gamepad detected, hold a button on your device to configure"

Pressing the buttons doesnt do anything.

Tried a few times but to no avail.

Also, I dont see the discord chat channel when I try to discuss this on discord. I can only see announcements, where I cannot chat.

sertiana, 2021-07-27:
You have to give a reaction to the announcement post in order to get the full access of the server. If it doesn't work, leave and try to rejoin once again.

"No Gamepad Detected" is a very common issue within rg351p/m devices. While it's most likely a hardware problem, it can also be caused by low battery power. Recharge your device and try again.

closed 2021-07-27 by dhwz

johnnyzen, 2021-07-27:
Great. Just by plugging the device in to charge, its fixed it. Thank you for your time :)