AmberELEC/AmberELEC #599: Software Failur on SNES and Sega Genesis

KnallerGer, 2021-08-28:
when I start a SNES or a Sega Genesis game the error message appears as shown in the picture.  Playstation 1 games run flawlessly.  until 2 days ago the same SNES games were still running without errors.

1. Go to SNES Games
2. Click on any Game
3. See error

[image]

I hope someone can help me and thank you in advance.

closed 2021-08-28 by dhwz

joseMartinezDev, 2021-08-31:
Why was this issue closed? I'm having this same problem with some roms. Is there any way to fix it?

dhwz, 2021-08-31:
Issues are not intended for support. "Bug report, only open after discord report!"
Join Discord for support, as this is no issue but a user misuse.