AmberELEC/AmberELEC #612: ParallelN64 incorrect overmapped binding with retrorun

nnolan-open, 2021-09-10:
### Describe the bug
when using ParallellN64 under, The R button is over-mapped with the start button. I believe that is b6(?) but don't hold my feet to the fire

In a certain favorite adventure game, this makes using my shield impossible; along with other strange pause menu behaviors. 
due to the "what you see is what you get" nature of Retrorun, I can't find where to change the defaults in ParallelN64 itself. The remap files that I could find, just reference device type "1" for joystick etc.

  Another thing to note is that the c-stick is not working. The c-toggle L2(?)  works, so that is probably less pressing than the overmap.
  
  ### To Reproduce
  - using /storage/.config/OS_VERSION
  20210603
  - set  to Retrorun/ParallelN64
  - go get your shield, and try to use it. Or try to pause then unpause.
  This caused some issues with other games, but the one i'm referring to is is  the best test-case. 
  
  ### Expected behavior
  Expected N64 R button to be mapped to an independent button, or somewhere to edit the autoconf respected by ParallelN64
  
  ### Screenshots
  n/a
  
  ### Device (please complete the following information):
  20210603 on a 351M
  
  ### Additional context
  one would expect that if it's neighbor, Retroarch/libretro would still reference the same autoconf provided by ParallelN64 (at least until overwriting); but they seem to have their own defaults/autoconf and bugs of their own.
  
  This probably also holds true for ParallelN64-GLN64

dhwz, 2021-09-10:
There is no mapping at all for retroron.

closed 2021-09-10 by dhwz

nnolan-open, 2021-09-10:
I think you may misunderstand. It's the actual hardcodes that are mis-mapped. Where are those located pre-build. Just try to follow the reproduce