AmberELEC/AmberELEC #623: Lines at bottom of the screen

TheCakeGit, 2021-09-24:
351ELEC - 20210603 Crazy Hedgehog (Hotfix3) and Anbernic 351V.

After about an hour or so I quite often get horizonal faint lines from the bottom of the screen when playing games which gets progressively worse. It looks like a heat issue, but I suspect it's a corrupted memory display issue.

Are you aware of this?

Thanks..

dhwz, 2021-09-24:
https://github.com/351ELEC/351ELEC/pull/585/commits/7f5a1b35adc37dc086fe4d98912d81edf5079355

closed 2021-09-24 by dhwz

TheCakeGit, 2021-09-29:
Thx for info!
I'll wait it out for the next full release. Not sure if beta is upgradable and not sure if there is a release package.

Rolen47, 2021-09-29:
> 
> 
> Thx for info! I'll wait it out for the next full release. Not sure if beta is upgradable and not sure if there is a release package.

Beta is upgradable and the releases are here:
https://github.com/351ELEC/351ELEC-beta/releases