AmberELEC/AmberELEC #626: Drastic Microphone

RyuuSlayer, 2021-09-26:
### Drastic cannot make use of Button mapped Microphone usage.
While there is a option to map the "Fake Mic" to a button, this doesn't actually work ingame. I also tried a type-c interface connected to a rode to try this option because i read that drastic does actually support real and fake mic, the interface doesn't require additional drivers so i figured it works (it doesn't), im not well versed on the technical side but mic support seems to be deactivated all together. 

### Is your feature request related to a problem? Please describe.
Causes progression problems in alot of DS games, Phantom Hour Glass at 4 points in the game, spirit tracks whenever you need to play the flute, bomberman land touch => all the minigames that require microphones, you get it, tons more.

### Describe the solution you'd like
While not everyone has a microphone setup worth more than the actual device, i don't think it's a bad idea to look into why button mapped Fake Mic functionality doesn't work. I replaced L2 with the Fake Mic because both L2 & R2 are used for swapping screens, so we don't actually loose anything.

### Additional context
Maybe someone has looked into this before and it is currently unfixable, but i wanted to drop this here anyways.

dhwz, 2021-09-30:
I'm sorry but L2 and R2 have different functions, L2 is switching the screens, R2 is switching between dual and single screen view, so both are needed.

closed 2021-09-30 by dhwz

RyuuSlayer, 2021-10-01:
okay, so leave 40% of games unplayable then, makes sense. you could also press in the left thumbstick.