AmberELEC/AmberELEC #641: Cannot access Drastic settings

thegriffindude, 2021-09-29:
### Describe the bug
There is no settings menu for Drastic.

### To Reproduce
Steps to reproduce the behavior:
1. Boot up a DS game
2. Try a lot of combinations that could be the settings combo and see that nothing comes up

### Expected behavior
A menu to pop up that contains save state, control, screen orientation settings.

### Device (please complete the following information):
 - Version of 351ELEC for RG351v (351ELEC-RG351V.aarch64-20210603)

### Additional context
Hey there folks! I'm having some issues accessing the settings on Drastic (RG351v).. First, Drastic was not installed so I went to Downloads > Packages to try to get it and it never downloaded it just kept saying an error has occurred. So I downloaded "drastic-v" from github and ran the install.sh. Games run flawlessly! but the screens are on top of each other and its not the best setup. How to I bring up a config menu to change this? I've tried a lot of button combos already and nothing seems to work.

dhwz, 2021-09-30:
Hm I don't understand what you mean? The menu of drastic? Can be accessed by pressing Fn (on V) / L3 (on P/M/MP).

closed 2021-09-30 by dhwz

thegriffindude, 2021-09-30:
> Hm I don't understand what you mean? The menu of drastic? Can be accessed by pressing Fn (on V) / R3 (on P/M).

This is what I mean.
https://streamable.com/jx4op6

dhwz, 2021-09-30:
Does/did the F button ever work for you anywhere else outside of Drastic?
Can you open RetroArch Menu with L3+F?
It's definitely working in Drastic with the F button I tried it myself.
So a) your button is broken or b) you messed up your drastic config as you said you've installed it manually then maybe wrong

thegriffindude, 2021-09-30:
Yes the F button works fine in Retroarch, just not in Drastic for me.

Heres me attempting to download Drastic the normal way: https://streamable.com/kadvhq

I downloaded the "drastic-v" and "drastic" folders from here: https://github.com/351ELEC/packages
I tried to install one, uninstalled that one when it didnt work, then installed the other and still nothing.

thegriffindude, 2021-09-30:
Are you able to send me all your Drastic files/configs? That'd be awesome

thegriffindude, 2021-09-30:
> It's definitely working in Drastic with the F button I tried it myself. 

Are you able to send me all your Drastic files/configs? That'd be awesome

dhwz, 2021-09-30:
try to use on of our beta releases drastic is already part of it