AmberELEC/AmberELEC #712: Amiga Mouse automatic moves to border

kangoroo82, 2021-11-15:
In uae the mouse curser moves automatic to the border.

YonKuma, 2021-11-15:
@kangoroo82 What device (P/M/V/MP) are you using, and what version of the OS (stable or beta and if beta what release)?

kangoroo82, 2021-11-15:
351v and beta-20211111_2216

YonKuma, 2021-11-15:
Thanks. I'll look into it

closed 2021-11-21 by dhwz

kangoroo82, 2021-11-23:
thanks for your work, but the mouse curser always moves from the top of the bottom of the screen, till it reaches the boarder.

YonKuma, 2021-11-23:
@kangoroo82 You're using the latest release? Are you using puae as your emulator, or are you using a non-default emulator? Is there any chance you have overrides in place that are overwriting the setting?

If you get the chance, please open up the RA quick menu, go to Options > Input > Mouse Controller Deadzone (or something, don't remember the exact name), and check that it's being set to 25% rather than 20%

kangoroo82, 2021-11-24:
ok. now it's great. thank you. i changed it manually. i thought with the latest beta it is done. but it wasn't changed. many thanks. please put that in the next beta.