AmberELEC/AmberELEC #721: Some PortMaster scripts are looking for a $directory folder

mikepruett3, 2021-11-23:
Just an update from my testing with the latest Pineapple Forest release, and PortMaster. I have found that a few of the "SH" shell scripts are expecting a $directory environment variable to be present at runtime. Looks like they need this var to point to /storage/roms/ in order to find the files.

I was able to add an export line to the beginning of the SorR.sh script in order to get Streets of Rage Remake to work...

```bash
#!/bin/bash

export directory=/storage/roms/   #<<THIS LINE HERE>>

if [ -d "/opt/system/Tools/PortMaster/" ]; then
  controlfolder="/opt/system/Tools/PortMaster"
elif [ -d "/opt/tools/PortMaster/" ]; then
  controlfolder="/opt/tools/PortMaster"
else
  controlfolder="/$directory/ports/PortMaster"
fi

source $controlfolder/control.txt

get_controls

export LD_LIBRARY_PATH=/$directory/ports/sorr/libs:/usr/lib32:/usr/local/lib/arm-linux-gnueabihf/
cd /$directory/ports/sorr
$ESUDO ./oga_controls bgdi $param_device &
./bgdi $(find "/$directory/ports/sorr" -type f -iname "sorr.dat")
$ESUDO kill -9 $(pidof oga_controls)
unset LD_LIBRARY_PATH
$ESUDO systemctl restart oga_events &
printf "\033c" >> /dev/tty1
fi
```

Could this Environment variable be added to the newer releases?

dhwz, 2021-11-24:
Wrong place we are not responsible for these scripts.
Contact @christianhaitian at our Discord server.

closed 2021-11-24 by dhwz

christianhaitian, 2021-11-24:
SorR was corrected yesterday afterI saw made aware.