AmberELEC/AmberELEC #723: No capcom games

QBall21, 2021-11-23:
Hello,

I updated my RG351M to the latest 351ELEC release, pineapple forest using the.tar image. All was well but after turning the console one I can't see any of my Capcon games. I checked the SD card and they are there, but the system doesn't see them.

Did anyone had this problem? Any ideas on how to sort it?

Thank you.

joshterrill, 2021-11-23:
If you check the latest release notes: https://github.com/351ELEC/351ELEC/releases/tag/20211122

I believe the capcom games got moved into the MAME section. See the above link for the details.

QBall21, 2021-11-23:
Thank you very much for your answer.

joshterrill, 2021-11-23:
@QBall21 if this answers your question and using the mame/fbn folder works for you, please close the ticket out. Thanks!

closed 2021-11-24 by QBall21