AmberELEC/AmberELEC #725: eduke rom directory doesn't have a system in es_systems.cfg

TheRobotFactory, 2021-11-26:
Personally I think doom, ecwolf and eduke really belong under ports but they are in the root. The eduke folder never shows up because it doesn't have a system in es_systems.cfg. 

There are some other es_systems inconsistencies where the system/directory/theme names are not all the same which wakes things a little confusing for theme creators, it's maybe worth looking at that while you are adding the eduke system in?

dhwz, 2021-11-26:
It's not a inconsistency at all, Doom and Wolfenstein are handled like a real system.
And also Duke Nukem is a included in a system, it's named Build Engine (foldername build).

closed 2021-11-26 by dhwz

TheRobotFactory, 2021-11-26:
I'm just trying to help, I'm not just giving you my opinion and expecting you to agree. 

The 351Elec image creates an eduke folder in GAMES. There is no eduke system in es_systems.cfg
The 351Elec image creates a build folder in GAMES. There is no build system in es_systems.cfg

These are the inconsistencies I was reporting as an issue. Put something in those folders, get nothing on the ES menu. It's an error.

Whether a single game that only has a few mods should be classed as a system or not, is not an inconsistency, it's just a matter of opinion. I wasn't reporting it as an error.

The other inconsistencies I was mentioning were things like :-

PCengine CD has a name of pcenginecd, a path of pcenginecd and a theme of pce-cd... inconsistent.
Neogeo CD has a name of neocd, a path of neocd and a theme of neogeocd... inconsistent. 
Final Burn Neo has a name of fbn, a path of fbneo and a theme of fan... inconsistent. 
 
There are also things like not including a folder for Sega Mark III (the jp version of the master system) but including a folder for famicom (the jp version of the NES). That's inconsistent too.

None of these affect the usability of the software so I wasn't reporting them as issues. Its a shame to see this issue closed without a fix because no-one bothered to read it before closing it. I even put the error right in the title so it wouldn't get missed in the body text :(