AmberELEC/AmberELEC #733: Down or up button isn't released

dfilskov, 2021-12-01:
If you're scrolling up or down in a game list and use the analogue stick to switch platform (right or left) the up or down button is held down (even if you've released it) when you retun to that game list.

dhwz, 2021-12-02:
can you please explain that a bit more in detail as I don't know what you mean

dfilskov, 2021-12-02:
Perhaps you can see the bug here?

https://www.dropbox.com/s/99ind0y4qj4s55p/menu_button_not_released.mp4?dl=0

Sorry for the bad video quality.

dhwz, 2021-12-02:
Can't reproduce that here on my device. You've used a MP right? I know that some devices have issues with the DPAD where the buttons sometimes stuck, but thats not related to our software.

dfilskov, 2021-12-02:
The D-Pad works flawlessly - buttons never stick - only in the menu when I do what I do in the video.

Do you try to repro this way?

Open e.g. the C64 game list.
Hold DPad down to scroll down the game list.
While holding down press the right analogue stick to the right shortly (the menu goes to the next core/platform)
Release DPad down.
Press the right analogue stick to the left to go back to the C64 game list.
Watch as the menu thinks you're stil holding down the DPad down button.

dhwz, 2021-12-03:
Yes that's what I tried, it doesn't happen on my device.
As said that's a known issue of the hardware, the pressed DPAD buttons sometimes stuck (I didn't mean the button physically)

dfilskov, 2021-12-03:
Hmm that's odd.

If it's the hardware I guess I should be able to repro in other situations.

I've tried the combinations that "stick" in
 both the game settings menu (opened by pressing Start) and in 351Files and in the PPSSPP menu. But I can't repro there. Only in the game lists.

dhwz, 2021-12-08:
Please go to the Retro Game Handhelds Discord #rg351mp there is a user which can explain how he fixed his broken device.
I'll close the issue now as this is not related to our image.

closed 2021-12-08 by dhwz

dfilskov, 2021-12-08:
Thanks. I will ask him/her! - Are we sure my decide is broken even when I can't repro in any other menus/software but consistently repro in the Emustation game lists?

dhwz, 2021-12-08:
Not 100% sure but it sounds similar to his issues.