AmberELEC/AmberELEC #734: ScummVM Issues after Update to Pineapple

Ronner51, 2021-12-05:
I play a lot of the good old Adventures with the ScummVM Engine, and had no Problems with ScummVM on the Hedgehog Version.

In the Update to Pineapple is a newer Version of ScummVM, but I can't:

-Use the Modern Theme
-Use the Extras, some games give me a Message at the start that some Extra files missing
-Can't set the Quick-Menu in the Controll-Settings

The Extra & Theme Folder is configured to: Roms/gamedata/ScummVM/(Theme or Extra)

I've tried to delete the Settings "INI" file, and all the Settings were gone, but it doesnt help for that problem.

The Update to Pineapple is a great Step Forward, but please bring me ScummVM in the Modern Version back. :)

dhwz, 2021-12-05:
You are using the wrong emulator, the RetroArch core never had modern theme, you've to use scummvmsa which is the default emulator.

Ronner51, 2021-12-05:
Hi Sir, thanks for the Answer but i do use the standalone ScummVM Emulator.

Ive tried it also with the Retroarch Core but as you say the Extra Things didn't work there.

dhwz, 2021-12-05:
I think you probably should delete your scummvm.ini and start from scratch, as it works perfectly here.

Ronner51, 2021-12-06:
I have tried to delete it, but it bring the same problem.
I think I have to do a completly new install.

Thank you anyway.

dhwz, 2021-12-07:
I'll close this issue, if necessary we can reopen it.

closed 2021-12-07 by dhwz

Ronner51, 2021-12-09:
So I formated my SD card and made a complete new install of Elec Pineapple.
I can not set my ScummVM standalone with the modern theme like it worked on the Hedgehog Version, even when I use the same Theme and Extra Folder that worked on the Hedgehog Version.

dhwz, 2021-12-09:
Why are you modifying the theme and extra folder?
The theme is already in the image itself and works already if you start it for the first time.

Ronner51, 2021-12-09:
I have tried it, there was no folder configured and I didn't find the folders somewhere After the fresh install.

dhwz, 2021-12-09:
Because you don't need to configure any folder, the theme is installed on it's default path
And the extras are compiled into the binary

Ronner51, 2021-12-09:
When I make a completly fresh install and open the ScummVM over the Tools-Section without adding any Files or change any Configuration, the only Theme I can choose is the Classic Theme.

dhwz, 2021-12-09:
Sorry I can just repeat myself if you start it with a clean install it is already starting with the remastered theme, not with classic.
No other themes are available. I tried it multiple times and it works as expected.

Ronner51, 2021-12-09:
Hmm thats strange, maybe i'll make a new install when a new version is released.

Anyway thank you for the help. I like 351ELEC very much and use it every day, keep the good work going sir ;)