AmberELEC/AmberELEC #872: Question Regarding Saving In 351Elec (AmberElec)

SamuelBanya, 2022-06-08:
NOTE: I don't use Discord, thanks.

I have a question:
What Linux user owns the '/storage/.config/retroarch' user directory?

Is it a dedicated 'retroarch' or 'rg351mp' user, etc?

Asking since I'm trying to modify the saves folder to be an alternate folder to be '/storage/saves' but it doesn't seem to allow it to actually be saved.

'/storage/.config/retroarch' seems to be a symlink only known to Anbernic devices as well.

dhwz, 2022-06-08:
Use bind mount if you want to use an alternative folder, but don't change the real one that will probably break a lot of things in the image.

closed 2022-06-08 by dhwz

SamuelBanya, 2022-06-08:
Hey @dhwz what do you mean by 'bind mount'?

I ask because I can't just transfer saves with '/storage/.config/retroarch' directory's default save directory because its a symlink only known to Anbernic devices. This the Catch 22 that I'm aiming to resolve and need some help with this.

Would you be able to state a workflow in terms of using an alternate directory of even transferring the saves between two Anbernic devices (RG351MP and RG351V) as well as Batocera Linux, all of which use Retroarch?

I ask because I think this is a weird loop hole since they all use Retroarch and should be able to have saves intermingle between them all. I know the 'rcloud' option exists but I don't use Google Drive either, so I wanted to know how to just transfer saves manually (which I was doing with related 'rsync' commands).