AmberELEC/AmberELEC #886: [meta] discord-to-somethingelse bridge?

gabuzom, 2022-07-07:
Sorry for opening a meta-issue, and especially WITHOUT going to Discord(tm) first.

First of all i want to praise the wonderful work on AmberELEC! you people are truly pushing these devices in the pantheon of retro-gaming! The experience is amazing! <3

Now: I, like many, consider Discord(tm) as being grossly irrespectful of privacy, starting by their policy to request authentication via email and phone# (a policy that is discriminatory of people without phone numbers...).

I am not judging or blaming people who choose to use Discord(tm) here, but feel incapacitated and discriminated against, as I won't (for ethical reasons) join such a platform, and therefore cannot respect your usage of asking to Discord(tm) first.

Hence my suggestion: could you people deploy a discord-to-matrix bridge, or any other solution that would enable users who are not captured in Discord(tm) to interact with you? It seems like it would benefit everyone....

There are several options for that:
https://github.com/matrix-org/matrix-appservice-discord
https://matrix.org/docs/projects/bridge/matrix-appservice-discord/
https://bitlaunch.io/blog/how-to-bridge-discord-and-matrix/

Cheers, and thanks again fro the amazing software!

gabuzom, 2022-07-24:
https://matrix.org/docs/projects/bridge/matrix-appservice-discord 
can be self-hosted but https://t2bot.io/discord/ runs it for free

Sosukodo, 2022-08-01:
I have a serious bug to share but will not jump through your (discord) hoops to report it.

medeirosT, 2022-09-28:
Would an IRC server be okay?

gabuzom, 2022-09-29:
I guess it would work in oder to not be "captured" in the Discord(tm)... but am not sure how easy it is to interconnect Discord(tm) and IRC... 

The advantage of Matrix is its extensive bridge. In a same channel could interact users of Discord(tm), Matrix, IRC, Telegram(tm), etc...

https://matrix.org/bridges/

Thanks for considering our request though!

Some0nee, 2022-09-29:
It is possible to bridge discord and irc via a bridge bot, the bot would
conglomerate both platforms into one chatroom, relaying the messages back
and forth between platforms which isn't hard at all. The problem is hosting
the bot.

Sosukodo, 2022-09-29:
Forgive me if I appear daft but what's wrong with *discussions* and *bug reporting* on GitHub? Why the need to go anywhere else? I don't get it.

dhwz, 2022-10-04:
@gabuzom
He wasn't talking about a connected server, just a separate channel on a IRC server (preferably on Libera.Chat).

medeirosT, 2022-10-04:
An AmberELEC channel was created in libera.chat for those who don't agree with Discord's terms of service.

Join us at #amberelec ! :)

closed 2022-10-04 by medeirosT