AmberELEC/AmberELEC #908: [Feature Request] include pygame into the build

Dreagonmon, 2022-10-24:
This console is powerful, it must have no problem using the SDL2 based pygame.

I have written [a text reader using pygame](https://github.com/Dreagonmon/rs97_TextReader), and it can run on RS-97 and RS-90 devices. the system RetroFW and OpenDingux they all have pygame as the builtin python library. Now I hope to port it to RG351P.

There are lots of games written in pygame: https://www.pygame.org/tags/all
Many of them are opensourced, it would be easy to port to this console if pygame is supported.

Thanks to vscode remote - ssh, the games written in python can be debuged directly on the console.

Cebion, 2022-10-24:
Or you package them up and add them here https://github.com/christianhaitian/PortMaster/

Dreagonmon, 2022-10-24:
I'm not familiar with the C/C++ projects, but I will try it.

closed 2022-10-24 by Dreagonmon