AmberELEC/AmberELEC #913: Autostart.sh homebrew entry when should be ports

Rocky5, 2022-11-03:
## This is only for actual bugs that can be reproduced on multiple systems!  ##
 For Support visit us in our Discord
https://discord.com/invite/R9Er7hkRMe

### Getting started
Before opening an issue, check the release notes of the version that you are using and recent commits.  Be sure to fill out the template completely, accurately, and with as much detail as possible.  Issues without sufficient detail will be rejected and closed.

### Describe the bug
A clear and concise description of what the bug is.

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
If applicable, add screenshots to help explain your problem.

### Device (please complete the following information):
 - Version of AmberELEC that you are using

### Additional context
Add any other context about the problem here.

dhwz, 2022-11-03:
There is absolutely nothing wrong here, the two games in homebrew are hidden (because they are just for internal use) in gamelist.xml thats why they are not visible, if you want them be visible you've to remove the hidden flag from gamelist.xml.

And they don't belong in ports, ports is reserved for Portmaster usage.

closed 2022-11-03 by dhwz

Rocky5, 2022-11-03:
Ah ok my mistake, I didn’t realise I had to remove the hidden flags.