AmberELEC/AmberELEC #936: Pico-8 does not show up as a System

chrigi, 2023-01-07:
I was on the last 351ELEC version before AmberELEC and updated to the latest AmberELEC but now I don't see a Pico-8 System anymore.

I've followed the note in the release notes: "If you've used Pico-8 system before, remove pico8 and pico8_dyn from roms/pico-8 folder, add the pico8_64 and pico8.dat from the Pico-8 Raspberry Pi package (v0.2.4b or newer) to roms/pico-8 folder" and removed the mentioned files and replaced them with the `pico8_64` and `pico8.dat` files.

I also had 2 sh scripts in the pico-8 roms folder, to launch Pico-8 normally or with the pixel perfect settings as well as a corresponding gamelist.xml file poining to them. I've also updated the scripts with the one from here: https://github.com/AmberELEC/AmberELEC/blob/20220627/packages/amberelec/sources/pico-8.sh

But no matter what I do, the system does not show up. It's selected in the options to show.
I've also tried removing the scripts and gamelist file but that didn't help. I could launch it through the Tools menu though.

dhwz, 2023-01-08:
Unless you've put some games into the folder it won't show up.
.sh files in there are no longer supported, the launch script is in now tools.

chrigi, 2023-01-08:
Sorry for not mentioning that. Yes, there are games in there, in the `carts` sub-folder.
So now I have: `roms/pico-8/carts/*.png` and `roms/pico-8/pico8.dat`, `roms/pico-8/pico8_64`.
But still, pico-8 does not show up as it's own system.

Is there any way to modify that launch script? Previously I had a second script passing `-width 384 -pixel_perfect 1` so it won't start it scaled.

dhwz, 2023-01-08:
Can you please check emulationstations config files on /storage/.emulationstations maybe you've accidentally hidden the png extension.

dhwz, 2023-01-08:
And no I'm sorry the script can but be changed, but we properly could add a switch to ES config menu. I didn't even know you can change the scale.
But be also aware of you switch to pre-releasese you could also use fake-08 core where you can use all sorts of RetroArch functions.

chrigi, 2023-01-08:
Ah, there it is. In `/storage/.emulationstation/es_settings.cfg` png files were hidden: `<string name="pico-8.HiddenExt" value="p8;png" />`. I removed that line and now the system shows up :)

Thanks a lot for the help and sorry for posting about a misconfig here on the issue tracker. I really did not remember this at all. No clue why I set this.

Thanks for the hint regarding the pre-releases. I'm not sure if I'm going to switch, I'd rather have a solid version than one I have to constantly update. A switch to launch it without scaling would be very nice :) But if the next release will change this anyways then I'll figure it out using the fake-08 core then.

closed 2023-01-08 by chrigi