AmberELEC/AmberELEC #958: About the newly added simulator

clondy999, 2023-03-29:
I tried to separate the lns of openbor and transplant it to this project, and now it can run successfully, but the control cannot be used, can you help me?

dhwz, 2023-03-29:
I don't understand what you're trying to do.

clondy999, 2023-03-30:
Well, I mean I want to migrate the branch of openbor, v6630-fflns. I have successfully migrated and can run Final Fit LNS Ultimate. However, the controls for the 351V device are not available. I want to seek help

dhwz, 2023-03-30:
I'm sorry we can't give support for foreign software builds.
Controls won't work if it's not build for AmberELEC.

closed 2023-03-30 by dhwz

clondy999, 2023-04-10:
I was too busy last two weeks and didn't pay attention to your information
The source code address is: https://github.com/gonzalomvp/openbor
This is specifically written for Final Fit LNS Ultimate

clondy999, 2023-04-10:
I carefully read your information and found that I didn't explain it clearly. Well, I mean:
I have successfully ported the project on Amberelec, but the control is not available

clondy999, 2023-04-10:
@dhwz