AmberELEC/AmberELEC #960: About Retroarch Language Switching

clondy999, 2023-04-18:
The device I am using is 351V
The version used is: 98504482f319a8b50d31034bb4f352e7d8037c90
I added the retroarch language pack through SSH, which is Japanese and Traditional Chinese, and saved it to exit.
Entering retroarch again will result in a font error, displaying a chaotic font.

[image]

dhwz, 2023-04-18:
Wdym b you've added language pack?

clondy999, 2023-04-18:
yes. I have added the language pack.
This bug can be reproduced 100%.
If English is pre-set and then Chinese is set, the display will be normal. Then save the configuration file of RetroArch and exit.
Entering RetroArch again causes this issue.

dhwz, 2023-04-18:
Again I don't understand why you did add something, all languages are already included and your not using the latest release of AmberELEC.

clondy999, 2023-04-18:
This has nothing to do with language packs!
If the default language is Chinese, entering RetroArch will cause this issue

dhwz, 2023-04-18:
If it's still happening on the latest RetroArch version, then it's RetroArch's fault, sorry we can't do anything there.
So it's RetroArch which is saving wrong settings, you may need to report it there not to us.

clondy999, 2023-04-18:
Okay.
I can only fix it from the encoding

dhwz, 2023-04-18:
https://github.com/libretro/RetroArch/issues/12215

clondy999, 2023-04-19:
resolved

closed 2023-04-19 by clondy999

dhwz, 2023-04-19:
Can you tell what you did? Maybe we can add it to our docs?

clondy999, 2023-04-19:
Certainly. I chose to change the theme of retroarch
I have tried the link method you provided, but adding RetroArch fonts to the path also fails

clondy999, 2023-04-19:
There are no issues with the XML theme

dhwz, 2023-04-19:
Ok thanks

clondy999, 2023-04-19:
This is what I should do