AmberELEC/AmberELEC #967: about retrorun

clondy999, 2023-07-25:
I want to ask for help, how to change the control buttons of retrorun.

dhwz, 2023-07-25:
you can't the buttons are hardcoded

closed 2023-07-25 by dhwz

clondy999, 2023-07-25:
@dhwz 
I understand that this is hard coding
May I ask if you could tell me. How should I make the changes
I can't find the specific location

dhwz, 2023-07-25:
The button configuration is hard coded into the retro run binary, there is currently no way to change the configuration. There is no config file etc.

clondy999, 2023-07-26:
@dhwz 
I have some knowledge of hard coding
At present, I want to print my key values on retrorun, but I can't find a specific location

navy1978, 2023-07-26:
As dhwz said the keys are hardcoded, I'm not sure I've understood what you want to do, in any case if you want to swap the L1 and L2 triggers with R1 and R2 just wait for the next pre-release, this will be possible. 
For the future we can take in consideration to add this possibility via the cfg file.

clondy999, 2023-07-26:
I'm not responding to the direction and START buttons right now. I don't know why.
Then I can only wait for the new version to be released