AmberELEC/AmberELEC #1027: Upstream `gamecontrollerdb.txt` mappings

offalynne, 2024-12-05:
Is there a compelling reason `gamecontrollerdb.txt` changes are not upstreamed to the source repo https://github.com/mdqinc/SDL_GameControllerDB ?

I suppose if there are udev drivers or other changes that make AmberELEC different than a generic distro this might cause compatibility conflict but surely this is not the case for all handheld device controls ? I think it would be ideal for as many mappings as possible to move upstream

dhwz, 2024-12-05:
Well that's actually not possible the button mappings are not identical across different distros. And also e.g. the button hardware order in AmberELEC is different to others.

offalynne, 2024-12-05:
> button mappings are not identical across different distros

then why do you ship the db in the first place ?

dhwz, 2024-12-05:
Wdym? The gamecontrollerdb is required for several SDL related games and emulators.

offalynne, 2024-12-05:
> *"gamecontrollerdb is required for several SDL related games and emulators"*

then obviously it stands to reason there should be oone authoritative source so that all SDL related projects can share the same mappings

> *"why do you care"*

as a maintainer of the project it's obviously in my interest (and all your users',
arguably) that the data is correct upstream so there are as fewer issues and greater compatibility for projects future and past.

dhwz, 2024-12-06:
> your use of the db obviously refutes the suggestion that the data should be different per distro -- the whole point of having the db is that it aligns hundreds of existing devices to the same mapping standard

Seems you still don't get it, these hundreds of devices are completely irrelevant for us (I could throw out any of them and just keep the two controllers added by us). The only thing that matters is we define a identical mapping across several devices which our OS supports.
The buttons may (and are) differently aligned on other handheld distros.
You are NOT enforcing us to use a different mapping (e.g. your Go-Super Controller is incorrect according to our standards).
Also it's not as simple as you might think we don't just use SDL and gamecontrollerdb we've to keep mappings identical across several other controller setups, e.g. RetroArch, EmulationStation which don't use SDL and gamecontrollerdb.

offalynne, 2024-12-06:
I understand well that your mappings naively change what would be a simple thing to fix systemwide with, for example, competent udev rules.

> *The only thing that matters is we define a identical mapping across several devices which our OS supports*

Indeed, you trade broad open source support for exclusive and confused support for your own software

> *buttons may (and are) differently aligned on other handheld distros*

There is only one canonically correct mapping, perhaps it's not possible for you to understand that this is the very simple basis of the db itself

> *You are NOT enforcing us [..] a different mapping [..] your Go-Super Controller*

Indeed you are free to do as you please, whether that includes contributing to what you are using, or as in this case, selectively misapplying data for your own misuse. The Go-Super mapping was authored by christianhatian, I'm sure he can help you work that out for yourself.

> *I could throw out any of them and just keep the two controllers added by us*

You should certainly, go ahead and do that, then

> *we don't just use SDL and gamecontrollerdb we've to keep mappings identical across several other controller setups, e.g. RetroArch, EmulationStation*

Indeed, when you don't actually fix an issue (such as with a udev rule) but misapply downstream fixes, you're only creating more issues and future compatibility problems for other contributors and developers to detangle later

closed 2024-12-06 by dhwz